libimobiledevice/libimobiledevice #1629: Merge backups

Vraiment, 2024-12-08:
For "reasons", I have 4 different backups of the same device, 3 made with `idevicebackup2` and 1 with iTunes, which add up to ~250 GB, is it possible to merge them?

To provide a simplified version of the problem imagine I first backed up my device with iTunes and then for the next 3 days I ran: `idevicebackup -i backup --full $TIMESTAMP` and now I have something like:

-> iTunes/$UUID
-> 20241201/$UUID
-> 20241202/$UUID
-> 20241203/$UUID

I'd like to create a single backup out of these 4 versions

mexmer, 2024-12-08:
itunes doesn't provide such functionality, and neither provide ios.

you need only last one, which should be the one without date. if you need some content from previous, there is software like iPhone Backup Browser, or others, which can allow you to peek into backup files (even encrypted ... if you know backup password, that is) and copy out what you want.

mind when you do backup to computer in itunes, it always creates full backup.

AiXanadu, 2025-01-18:
Merging backups is a complex operation, and it is not possible to merge all data.
However, merging photos and music is still possible.
The process is to use SQLite to open the DB file, then find photos and music, and extract the corresponding files. Photos are SQLite. You need to open the two backup photo databases again, and then carefully merge the databases.
Add/modify the new database size and the merged photo path to the backup you want to restore. You need to extract the last item file of the backup, which is a plist, and then modify it. Save and restore.