libimobiledevice/libimobiledevice #1631: You are using macOS 15.

easylolicon, 2024-12-09:
brew install libimobiledevice

Warning: You are using macOS 15.
We do not provide support for this pre-release version.
It is expected behaviour that some formulae will fail to build in this pre-release version.
It is expected behaviour that Homebrew will be buggy and slow.
Do not create any issues about this on Homebrew's GitHub repositories.
Do not create any issues even if you think this message is unrelated.
Any opened issues will be immediately closed without response.
Do not ask for help from Homebrew or its maintainers on social media.
You may ask for help in Homebrew's discussions but are unlikely to receive a response.
Try to figure out the problem yourself and submit a fix as a pull request.
We will review it but may or may not accept it.

Error: unknown or unsupported macOS version: :dunno

iayusuf, 2025-01-06:
Have the same issue.

nikias, 2025-01-06:
For macOS, there is an easier way by using my build script:
```bash
mkdir -p limd-build
cd limd-build
curl -o ./limd-build-macos.sh -L http://is.gd/limdmacos
bash ./limd-build-macos.sh
```