libimobiledevice/libimobiledevice #1633: idevicepair error -5

Enekololo, 2024-12-14:
`ERROR: Device 00008110-001A30E63E89A01E returned unhandled error code -5`

i keep having the same error and could not find the solution pls help me

bclouthier, 2024-12-24:
Yes, I had the same issue. I could see the device but I could not pair the device or initate a backup. What worked for me on Fedora 41 was running the following command in the terminal, `sudo update-crypto-policies --set FEDORA40`. I'm running Fedora 41. Apparently, the crypto policy breaks the iOS devices ability to mount and share files and so you regress to Fedora 40's crypto policy. More info here, [https://bugzilla.redhat.com/show_bug.cgi?id=2321958.](url)

dennis-benzinger, 2025-02-03:
There's a less invasive workaround at:
https://github.com/libimobiledevice/libimobiledevice/issues/1606#issuecomment-2543116644

kfurkanbek, 2025-07-14:
Even though I agree with the solution suggested by @dennis-benzinger here,
I just downgraded, transferred and upgraded it back.

1. sudo update-crypto-policies --set FEDORA40
2. Restart
3. sudo update-crypto-policies --set DEFAULT