libimobiledevice/libimobiledevice #1637: Github Actions build does not contain every needed .dll

basketshoe, 2024-12-28:
I've compiled libimobiledevice via GH Actions but not every .dll is present. I've managed to find `libplist-2.0.dll`, `libimobiledevice-glue-1.0.dll` and `libusbmuxd-2.0.dll` but it still says it needs `libcrypto-3-x64.dll` and `libssl-3-x64.dll`.
How can I find those dlls?

timleg002, 2025-01-14:
Those are DLLs provided by OpenSSL. I fixed this by downloading MSYS2 and installing the OpenSSL package (`pacman -S mingw-w64-x86_64-openssl`)

basketshoe, 2025-01-14:
Thanks, I've found them

closed 2025-01-14 by basketshoe

ZhouLixuanwu, 2025-02-02:
> I've compiled libimobiledevice via GH Actions but not every .dll is present. I've managed to find `libplist-2.0.dll`, `libimobiledevice-glue-1.0.dll` and `libusbmuxd-2.0.dll` but it still says it needs `libcrypto-3-x64.dll` and `libssl-3-x64.dll`. How can I find those dlls?

Hi, I wonder how did you find these files?
`libplist-2.0.dll`, `libimobiledevice-glue-1.0.dll` and `libusbmuxd-2.0.dll` but it still says it needs `libcrypto-3-x64.dll` and `libssl-3-x64.dll
I checked the gh actions's build, but all I got is libimobiledevice-1.0

basketshoe, 2025-02-02:
@ZhouLixuanwu 
libcrypto and libssl can be found by installing [openssl](https://slproweb.com/download/Win64OpenSSL-$underscoreVersion.exe)

The other ones can be downloaded by clicking on any latest build, then scroll down and download the zip for your PC. You need to be logged in GitHub to download those artifacts
https://github.com/libimobiledevice/libplist/actions
https://github.com/libimobiledevice/libimobiledevice-glue/actions
https://github.com/libimobiledevice/libusbmuxd/actions