libimobiledevice/libimobiledevice #1638: iOS 17 Development Path / Proposal

nanoscopic, 2024-12-31:
Is there realistically anyone who either is or intends to work on supporting iOS 17 in libimobiledevice?

I ask this as my assumption is that the answer is no, and I have a proposal for actually adding iOS 17 support and more to libimobiledevice.

I am the author of iosif ( https://github.com/dryark/iosif ), and the co-author of go-ios ( https://github.com/danielpaulus/go-ios )

iosif is the closest thing imo to being able to support iOS 17 ( in C code ), and I am probably the only person experienced with making iOS 17 work and the underlying protocols etc who would be interested in doing it in C.

My proposal is to change the iosif license to something compatible with libimobiledevice and integrate the portions of it needed and then extend it for iOS 17.

I have a few questions though that I need some answers to before being willing to do so:

1. Would it be possible to use an AGPL ( AGPL 3.0 ) license for such? I'm not a fan of companies being able to extend code and integrate it into products without contributing back. GPL 3.0 and AGPL 3.0 code can be intermixed but libimobiledevice appears to be on GPL 2.0. Can libimobiledevice be moved to GPL 3.0?

2. Who else if anyone can / would help me in the endevour? It would/will require significant effort.

3. Would it be acceptable to use the custom C archiver I wrote for nskeyedarchiver? There is an archiver in the libimobiledevice code already but it does not support all the archiver features that are required.

4. Would it be acceptable for me to spearhead such an effort?

mexmer, 2025-01-03:
could you be more concrete what doesn't work for you in iOS 17, because iOS 17 is supported except idevicedebug functionality. you need to use HEAD revision and not latest tag.

nanoscopic, 2025-01-03:
A wide variety of things ceased to function through the old non-authenticated mechanisms with iOS 17.

There are so many I can't even recall exactly which offhand.

One example is that a lot of values no longer report out through gestalt; if even any gestalt can be accessed any more through libimobiledevice?

Additionally there is a ton of functionality that was just never implemented in libimobiledevice tools. That's why I wrote iosif in the first place and paid Daniel to create and then continue working on go-ios.

You've never been able to start xctests using libimobiledevice for example.

So my answer would be similar to your question. The answer is "besides everything?"

nikias, 2025-01-03:
I am working on support for it (remoted to be precise), but it's not as easy as you might think. On macOS, we need to talk to the existing remoted because we cannot coexist with another solution; the device is really picky and only accepts a single connection.
It is implemented in pymobiledevice already. 
Also, gestalt access is dead...

nanoscopic, 2025-01-03:
It's false that you cannot coexist with another solution, and also false that you need to use the existing remoted utun.

You can pause remoted for a very short amount of time ( < 100ms ) just when needed. This works fine. I do it currently. It allows xcode to continue functioning while still letting you create a new utun instead of reusing the existing one.

I'm aware of how pymobiledevice works. It was implemented based on my and Daniel's codebases.