libimobiledevice/libimobiledevice #1639: Dont know how to make backup

MustarEske, 2025-01-01:
I want to make a backup of my iphone, but i dont know how these things work. My Macbook is Sequia 15.2 my iphone is 18.2. if someone could just make a complete guide from start to finish that would be very helpfull.

HanXin9527, 2025-01-03:
> I want to make a backup of my iphone, but i dont know how these things work. My Macbook is Sequia 15.2 my iphone is 18.2. if someone could just make a complete guide from start to finish that would be very helpfull.

如果只是要备份的话,用官方的iTunes就好啊

mexmer, 2025-01-03:
you don't need this project to backup your phone to your mac, just follow apple userguide
https://support.apple.com/guide/iphone/back-up-iphone-iph3ecf67d29/ios

closed 2026-09-24 by MustarEske