libimobiledevice/libimobiledevice #1648: Call idevicebackup2 twice, but timeout the second time

tryhand, 2025-02-28:
The first time, by adding a password to the phone (CMD_CHANGEPW), it worked
The second call (CMD_BACKUP) returns service client new failed, error -5 in mobilebackup2_client_new. But in the middle of the two intervals of ten seconds, it can be successful, is I did not release the service?

mexmer, 2025-02-28:
it would be better to show code that is failing.

nikias, 2025-02-28:
Backup service has an instance limit, but it is 5, so I don't think it's that. The weird thing is that it's an SSL error, so we would need to understand what exactly is happening with the code you are trying to use. Just show the code that is failing and maybe we can spot the problem.

tryhand, 2025-03-03:
Thank you for your reply.I customized the main function in the idevicebackup2 file and made it call repeatedly. The code for modifying the phone password and backing up are quite similar. 
The process of invocation is as follows: first, a function is used to set a password for the mobile device; then, a backup is called for; and finally, the password is removed. Currently, the invocation of the backup has timed out after calling mobilebackup2_client_new which follows lockdownd_start_service_with_escrow_bag.
Below is the overall code within the function. I omitted the processing for saving the file.

tryhand, 2025-03-31:
Thank you for your reply.I customized the main function in the idevicebackup2 file and made it call repeatedly. The code for modifying the phone password and backing up are quite similar. 
The current environment is Windows, VS2010, and OpenSSL 1.0.1c. iOS 18.3
The process of invocation is as follows: first, a function is used to set a password for the mobile device; then, a backup is called for; and finally, the password is removed. Currently, the invocation of the backup has timed out after calling mobilebackup2_client_new which follows lockdownd_start_service_with_escrow_bag.

Some logs
`
start check encryptend check encrypt
signal start
add device
lockdown device
Started "com.apple.mobilebackup2" service on port 55121.
Negotiated Protocol Version 2.1
Start Reading Info.plist
Start Open lock_sync
initial read failed!
start ExtractHash
done ExtractHash
start mobilebackup2_receive_message
DLMessageDownloadFiles
start mobilebackup2_receive_message
DLMessageDownloadFiles
start mobilebackup2_receive_message
DLMessageProcessMessageBackup encryption has been enabled successfully.
free mobilebackup2
mobilebackup2_client_free err:0
free device
initial read failed!
res_code:0. errcode:0
Creat Directory
start backup
signal start
add device
Backup directory is "D:\"
lockdown device
Started "com.apple.mobilebackup2" service on port 55131.
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
initial read failed!
Did not receive initial message from device!
version exchange failed, error -5
mobilebackup2_client_free err:0
tart ExtractHash
done ExtractHash
free afc
initial read failed!
initial read failed!
free device
initial read failed!
initial read failed!
initial read failed!
initial read failed!
res_code:-1. errcode:0
backup finish
`

Below is the overall code within the function. I omitted the processing for saving the file.
[backup2.txt](https://github.com/user-attachments/files/19046436/backup2.txt)
look forward to your reply @nikias @mexmer

nikias, 2025-03-31:
Use idevicesyslog and observe messages from backupd (you can use `idevicesyslog -p backupd -p BackupAgent2` to filter out only backup-related messages). I assume that after changing the password maybe the connection is just closed from the device side.

tryhand, 2025-04-01:
> Use idevicesyslog and observe messages from backupd (you can use `idevicesyslog -p backupd -p BackupAgent2` to filter out only backup-related messages). I assume that after changing the password maybe the connection is just closed from the device side.

this is change password and do backup logs.

`\win-x86>idevicesyslog -p backupd -p BackupAgent2
[connected:00008030-0018155822E2802E]
Apr  1 09:25:45 BackupAgent2(libMobileGestalt.dylib)[7783] <Notice>: Cache loaded with 5688 pre-cached in CacheData and 66 items in CacheExtra.
Apr  1 09:25:45 BackupAgent2(libMobileGestalt.dylib)[7783] <Notice>: elided platform fast path for key: re6Zb+zwFKJNlkQTUeT+/w
Apr  1 09:25:45 BackupAgent2(libMobileGestalt.dylib)[7783] <Notice>: elided platform fast path for key: nFRqKto/RuQAV1P+0/qkBA
Apr  1 09:25:45 BackupAgent2(libMobileGestalt.dylib)[7783] <Notice>: elided platform fast path for key: VasUgeSzVyHdB27g2XpN0g
Apr  1 09:25:45 BackupAgent2[7783] <Notice>: Device UDID: <private>
Apr  1 09:25:45 BackupAgent2(libxpc.dylib)[7783] <Notice>: [0x1035176b0] activating connection: mach=true listener=false peer=false name=com.apple.mobile.keybagd.UserManager.xpc
Apr  1 09:25:45 BackupAgent2(libxpc.dylib)[7783] <Notice>: [0x103517e50] activating connection: mach=true listener=false peer=false name=com.apple.mobile.usermanagerd.xpc
Apr  1 09:25:45 BackupAgent2(libxpc.dylib)[7783] <Notice>: [0x103517e50] invalidated because the current process cancelled the connection by calling xpc_connection_cancel()
Apr  1 09:25:45 BackupAgent2(libxpc.dylib)[7783] <Notice>: [0x5b60b4000] activating connection: mach=true listener=false peer=false name=com.apple.cfprefsd.daemon
Apr  1 09:25:45 BackupAgent2(CoreFoundation)[7783] <Error>: Not updating lastKnownShmemState in CFPrefsPlistSource<0x5b60a81e0> (Domain: com.apple.MobileBackup, User: mobile, ByHost: No, Container: (null), Contents Need Refresh: Yes): 0 -> 140
Apr  1 09:25:45 BackupAgent2(libxpc.dylib)[7783] <Notice>: [0x5b60b4100] activating connection: mach=true listener=true peer=false name=com.apple.lockdown.mobilebackup2
Apr  1 09:25:45 BackupAgent2(libxpc.dylib)[7783] <Notice>: [0x5b60d8000] activating connection: mach=false listener=false peer=true name=com.apple.lockdown.mobilebackup2.peer[80].0x5b60d8000
Apr  1 09:25:45 BackupAgent2(liblockdown.dylib)[7783] <Notice>: unlock_with_escrow: <private>
Apr  1 09:25:45 BackupAgent2(MobileKeyBag)[7783] <Notice>: <private>
Apr  1 09:25:45 BackupAgent2(libxpc.dylib)[7783] <Notice>: [0x5b60b4200] activating connection: mach=true listener=false peer=false name=com.apple.cfprefsd.daemon.system
Apr  1 09:25:46 BackupAgent2(Security)[7783] <Notice>: System Keychain Always Supported set via feature flag to disabled
Apr  1 09:25:46 BackupAgent2(libxpc.dylib)[7783] <Notice>: [0x5b60b4400] activating connection: mach=true listener=false peer=false name=com.apple.trustd
Apr  1 09:25:46 BackupAgent2(libxpc.dylib)[7783] <Notice>: [0x5b60b4600] activating connection: mach=true listener=false peer=false name=com.apple.trustd
Apr  1 09:25:46 BackupAgent2(libxpc.dylib)[7783] <Notice>: [0x5b60b4600] invalidated because the current process cancelled the connection by calling xpc_connection_cancel()
Apr  1 09:25:46 BackupAgent2(libxpc.dylib)[7783] <Notice>: [0x5b60b4700] activating connection: mach=true listener=false peer=false name=com.apple.trustd
Apr  1 09:25:46 BackupAgent2(libxpc.dylib)[7783] <Notice>: [0x5b60b4700] invalidated because the current process cancelled the connection by calling xpc_connection_cancel()
Apr  1 09:25:46 BackupAgent2(libxpc.dylib)[7783] <Notice>: [0x5b60b4500] activating connection: mach=true listener=false peer=false name=com.apple.iokit.powerdxpc
Apr  1 09:25:46 BackupAgent2(IOKit)[7783] <Notice>: Assertion feature: setting gAsyncMode to 1
Apr  1 09:25:46 BackupAgent2(libxpc.dylib)[7783] <Notice>: [0x5b60b4500] invalidated after the last release of the connection object
Apr  1 09:25:46 BackupAgent2(libxpc.dylib)[7783] <Notice>: [0x5b60d8140] activating connection: mach=true listener=false peer=false name=com.apple.mobile.usermanagerd.xpc
Apr  1 09:25:46 BackupAgent2(libxpc.dylib)[7783] <Notice>: [0x5b60d8140] invalidated because the current process cancelled the connection by calling xpc_connection_cancel()
Apr  1 09:25:46 BackupAgent2(SharedUtils)[7783] <Notice>: LA logging set up for this process.
Apr  1 09:25:46 BackupAgent2(LocalAuthentication)[7783] <Notice>: Creating LAContext new cid:1
Apr  1 09:25:46 BackupAgent2(libxpc.dylib)[7783] <Notice>: [0x5b60d8140] activating connection: mach=true listener=false peer=false name=com.apple.CoreAuthentication.daemon
Apr  1 09:25:46 BackupAgent2(LocalAuthentication)[7783] <Notice>: LAContext[1:1158:1419] created new cid:1
Apr  1 09:25:46 BackupAgent2(LocalAuthentication)[7783] <Notice>: evaluatePolicy:1013 options:{
} on LAContext[1:1158:1419] (async) cid:2
Apr  1 09:25:46 BackupAgent2(LocalAuthentication)[7783] <Notice>: evaluatePolicy on LAContext[1:1158:1419] cid:2 returned <NSError:0x5b6098690(com.apple.LocalAuthentication:-5) - {
    BiometryType = 2;
    NSDebugDescription = "Passcode not set.";
    NSLocalizedDescription = "\134U672a\134U8bbe\134U5b9a\134U5bc6\134U7801\134U3002";
}>
Apr  1 09:25:46 BackupAgent2[7783] <Error>: Skipping passcode policy evaluation: Error Domain=com.apple.LocalAuthentication Code=-5 UserInfo={NSDebugDescription=<private>, NSLocalizedDescription=<private>, BiometryType=2}
[... cut ...]
Apr  1 09:25:56 BackupAgent2(MobileKeyBag)[7785] <Notice>: <private>
Apr  1 09:25:56 BackupAgent2(libxpc.dylib)[7785] <Notice>: [0xd220b4200] activating connection: mach=true listener=false peer=false name=com.apple.cfprefsd.daemon.system
Apr  1 09:26:26 BackupAgent2(liblockdown.dylib)[7785] <Notice>: lockssl_handshake: <private>
Apr  1 09:26:26 BackupAgent2(liblockdown.dylib)[7785] <Notice>: Failed to enable SSL.
Apr  1 09:26:26 BackupAgent2(liblockdown.dylib)[7785] <Notice>: Could not check in with lockdown at '<private>': 1`

tryhand, 2025-04-01:
> after changing backup password keybag is reencrypted, if you have already backup on computer, it will also send you new manifest file. there will be new security context, therefore you can't change password and do backup in single connection. you need to disconnect after changing password, then connect to backup service again and do backup.

in my code,whether it's change password or backup, every time I use new idevice_t to connect,and in the end I free it.Is this considered as a single connection?

mexmer, 2025-04-29:
i can't show example of code, because it's commercial stuff, but in our software we first do check if backup is encrypted or not, if it's encrypted we just run backup, if it's not encrypted we set password, then do backup, then remove password.

we use only 1 device instance, and for either operation (change password, backup, reset password), we always create new lockdown client, service and backup client for each operation
```cpp
    idevice_t device = nullptr;
    auto dret = idevice_new_with_options(&device, udid,IDEVICE_LOOKUP_USBMUX);
    if (dret != IDEVICE_E_SUCCESS)
        return -1;
    bool ourpassword = false;
    bool backupdone = false;
    while(1) {
        lockdown_client_t lockdown = nullptr;
        auto lcret = lockdown_client_new_with_handshake(device, &lockdown, "OurTool");
        if (lcret == LOCKDOWN_E_SUCCESS)
        {
            uint8_t willEncrypt = 0;
            node_tmp = NULL;
            lockdownd_get_value(lockdown, "com.apple.mobile.backup", "WillEncrypt", &node_tmp);
            if (node_tmp) {
                if (plist_get_node_type(node_tmp) == PLIST_BOOLEAN) {
                    plist_get_bool_val(node_tmp, &willEncrypt);
                }
                plist_free(node_tmp);
                node_tmp = nullptr;
            }
            lockdownd_service_descriptor_t backupservice = nullptr;
            lcret = lockdownd_start_service_with_escrow_bag(lockdown, "com.apple.mobile.backup2", &backupservice);
            lockdown_client_free(lockdown)
            lockdown = nullptr;
            if (lcret == LOCKDOWN_E_SUCCESS) {
                mobilebackup_client_t backupclient = nullptr;
                mobilebackup2_client_new(device, service, &mobilebackup2);

                if (backupservice) {
                    lockdownd_service_descriptor_free(backupservice);
                    backupservice = nullptr;
                }
                // set ourpassword if there is nopassword
                if (!willEncrypt == 0 && !backupdone) {
                    // password set code here
                    mobilebackup2_client_free(backupclient);
                    backupclient = nullptr;
                    ourpassword = true;
                    continue;
                }
                if (!backupdone)
                {
                    // backup code here
                    mobilebackup2_client_free(backupclient);
                    backupclient = nullptr;
                    backupdone = true;
                    if (ourpassword)
                        continue;
                }
                if (ourpassword)
                {
                    // password removal code here
                    mobilebackup2_client_free(backupclient);
                    backupclient = nullptr;
                    ourpassword = false;
                    break;
                }
                if (backupclient) {
                    mobilebackup2_client_free(backupclient);
                    backupclient = nullptr;
                }
            }
            if(backupservice) {
                lockdownd_service_descriptor_free(backupservice);
                backupservice = nullptr;
            }
            if (backupdone)
                break;
        }
        if (lockdown) {
            lockdown_client_free(lockdown)
            lockdown = nullptr;
        }
    }
    idevice_free(device);
```

nikias, 2025-04-29:
You need to make sure to free up the mobilebackup2_client_t instance. Your logs show that the second time you try to connect, it fails to check in with lockdownd
```
Apr  1 09:25:55 BackupAgent2(liblockdown.dylib)[7785] <Notice>: unlock_with_escrow: <private>
Apr  1 09:25:56 BackupAgent2(MobileKeyBag)[7785] <Notice>: <private>
Apr  1 09:25:56 BackupAgent2(libxpc.dylib)[7785] <Notice>: [0xd220b4200] activating connection: mach=true listener=false peer=false name=com.apple.cfprefsd.daemon.system
Apr  1 09:26:26 BackupAgent2(liblockdown.dylib)[7785] <Notice>: lockssl_handshake: <private>
Apr  1 09:26:26 BackupAgent2(liblockdown.dylib)[7785] <Notice>: Failed to enable SSL.
Apr  1 09:26:26 BackupAgent2(liblockdown.dylib)[7785] <Notice>: Could not check in with lockdown at '<private>': 1
Apr  1 09:35:55 BackupAgent2(MobileDeviceLink)[7785] <Error>: lockdown_checkin_xpc timed out after 600s
```
But we don't really see the reason. Maybe it's just to quick when retrying? Anyway if you run idevicebackup2 tool manually it works, so something must still be "in use".

mexmer, 2025-04-30:
one reminder, as of ios 14 or something like that, you must confirm backup password change on device display, as of ios 17, you must confirm backup operation on device. it it not shown in my code, but there is callback before each operation, that displays warning window (depending on ios version), and it continues after uses cofirms it. so while it's not apparent from my code, there is always delay between each operation.

for older devices while callback are called, because we know device OS version, they are silently dismissed.

if backup service is waiting for password, it will not accept any other requests, but request timeouts i believe after some 60 seconds or so. it will then stop service and disconnect link.

tryhand, 2025-07-16:
> You need to make sure to free up the mobilebackup2_client_t instance. Your logs show that the second time you try to connect, it fails to check in with lockdownd
>
> ```
> Apr  1 09:25:55 [...]

You're right, there's something in use. I used idevicesyslog to look at the logs of other programs, and found that they all have an output indicating that the device is disconnected（<Error>: Computer disconnected: Lost connection）. How do I trigger this manually。

`Jul 16 14:28:02 BackupAgent2[2672] <Notice>: Drive returned unexpected error code: 1
Jul 16 14:28:02 BackupAgent2(MobileDeviceLink)[2672] <Notice>: 2672:6093156352|SocketStreamHandler.c:_SocketLogCallback| ERROR: SocketSend (data): error 22: Invalid argument
Jul 16 14:28:02 BackupAgent2(MobileDeviceLink)[2672] <Notice>: 2672:6093156352|StreamHandler.c:_WriteFunc| ERROR: Error sending data: Can't send data
Jul 16 14:28:02 BackupAgent2(MobileDeviceLink)[2672] <Notice>: 2672:6093156352|DeviceLinkConnection.c:_DLSendBytesAndDisconnectOnError| ERROR: Could not send data: 'Error writing length of data'
Jul 16 14:28:02 BackupAgent2[2672] <Error>: Computer disconnected: Lost connection
Jul 16 14:28:02 BackupAgent2(MobileBackup)[2672] <Notice>: Removing temporary directory at /var/tmp/backupd-x319j5yjnB5DnZo`

look forward to your reply @nikias