libimobiledevice/libimobiledevice #1650: It's probably something really obvious, but...

philthegitter, 2025-03-11:
...it's 2 in the morning here in Seattle and I'm too bleary-eyed to spot what I'm doing wrong.

I can't get idevicebackup2 to work.

I'm running Fedora 41, the xfce4 spin, on my trusty laptop 'xena,' an Acer A315-44P (AMD Ryzen 7 5700U, 32 GB of RAM). My phone is an iPhone 2020 SE, the miniature version of the iPhone 11.

Digikam can download photos and videos from this iPhone, it works flawlessly (except for large videos of around 6 GB, for which I use LocalSend). And the xfce4 file manager, Thunar, can see the iPhone and can mount its photos directory. I thereby have the ability to download photos and videos from the phone via Thunar as well as via digikam. Thunar shows the iPhone's photos directory as 

'gphoto2://Apple_Inc._iPhone_nnnnnnnnnnnnnnnnnnnnnnnX/202503__/' 

where the middle part is the iPhone's 23-digits-plus-one-capital-letter UDID (as shown by iTunes running under Windows 10) minus the hyphen between the eighth and ninth digits.

But what I want to do is download iMessage threads. I'm trying to create a backup on a 256 GB Samsung usb 3.1 thumb drive formatted as a single exFAT partition. I have created a directory on it, titled 'iphone'. Starting a terminal window from this directory, what happens is:

admin@xena:/mnt/exfat_usb/iphone$ idevicebackup2 -u 'nnnnnnnn-nnnnnnnnnnnnnnnX' backup --full .
No device found with udid nnnnnnnn-nnnnnnnnnnnnnnnX.
admin@xena:/mnt/exfat_usb/iphone$

I have tried running idevicebackup2 using both a hyphenated and an unhyphenated form of the phone's UDID.

The five apps libimobiledevice, libimobiledevice-devel, libimobiledevice-glue, libimobiledevice-glue-devel, and libimobiledevice-utils were downloaded via dnfdragora. All are the current Fedora version 1.3.0^20230705git6fc-41f5 release 5.fc41 for x86_64.

I'm going to bed. I'm hoping that when I wake up in the morning some kind soul will have pointed out what I'm doing wrong. Thanks in advance. Good night.

philthegitter, 2025-03-11:
The X is part of my UDID obfuscation. The real letter is E.

I get no success with idevice_id list:

admin@xena:~$ idevice_id list
ERROR: No device with UDID list attached.
admin@xena:~$

The above, even as I just used Thunar to download three videos from the iPhone, through a usb port, over the lightning cable. And then I used Thunar to delete those videos from the iPhone. A fourth video on the iPhone is 9.49 GB in size and so I'm using LocalSend right now to download it over wifi. For some unknown reason, files above a certain size, the threshold being somewhere around 5-6 GB, do not show up in the digikam and Thunar iPhone photos directory listing.

mexmer, 2025-03-12:
so you are accessing your phone not trough libimobile or ifuse, but most likely trough WPD or MTP protocol.

do you have usbmuxd installed on computer? because usbmuxd provides services necessary to connect phone in same mode as apple uses.

when accessing trough WPD/MTP only media are available to access, you can't do backup, or access some other stuff, like deviceinfo and such.

nikias, 2025-03-12:
The right command is `idevice_id -l` or just `idevice_id`.

philthegitter, 2025-03-12:
```
admin@xena:~$ idevice_id -l
ERROR: Unable to retrieve device list!
admin@xena:~$ idevice_id
ERROR: Unable to retrieve device list!
admin@xena:~$ 
```

```
admin@xena:~$ which usbmuxd
/usr/local/sbin/usbmuxd
admin@xena:~$ sudo systemctl start usbmuxd
[sudo] password for admin: 
admin@xena:~$ sudo systemctl status usbmuxd
× usbmuxd.service - Socket daemon for the usbmux protocol used by Apple devices
     Loaded: loaded (/usr/lib/systemd/system/usbmuxd.service; static)
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf, 50-keep-warm.conf
     Active: failed (Result: exit-code) since Wed 2025-03-12 12:05:52 PDT; 9s a>
   Duration: 18ms
 Invocation: 9914a4f422b8457598aed75c8d91e2a7
       Docs: man:usbmuxd(8)
    Process: 74279 ExecStart=/usr/local/sbin/usbmuxd --user usbmux --systemd (c>
   Main PID: 74279 (code=exited, status=1/FAILURE)
   Mem peak: 1M
        CPU: 9ms

Mar 12 12:05:52 xena systemd[1]: Started usbmuxd.service - Socket daemon for th>
Mar 12 12:05:52 xena usbmuxd[74279]: [12:05:52.341][3] usbmuxd v1.1.1-53-g91aa7>
Mar 12 12:05:52 xena usbmuxd[74279]: [12:05:52.342][0] Dropping privileges fail>
Mar 12 12:05:52 xena systemd[1]: usbmuxd.service: Main process exited, code=exi>
Mar 12 12:05:52 xena systemd[1]: usbmuxd.service: Failed with result 'exit-code>
lines 1-18/18 (END)...skipping...
(repeated until I Ctrl-C'ed it)
```

philthegitter, 2025-03-12:
> so you are accessing your phone not trough libimobile or ifuse, but most likely trough WPD or MTP protocol.

Sorry, you're over my head here.

Thunar did not see the iPhone until I installed the five libimobiledevice files listed above. But I also had digikam already installed. Fedora's digikam, from dnfdragora, however, kept crashing, so I compiled digikam 8.5.0 from scratch, from the git source, and it is totally stable on this machine.

philthegitter, 2025-03-15:
I've solved it. Life is good.

It seems that I was running into the crypto policies change issue which is being discussed at 
[](https://bugzilla.redhat.com/show_bug.cgi?id=2321958)

And so, issuing the command

`$ sudo update-crypto-policies --set FEDORA40`

and then rebooting, made the magic happen, as witness:

```
root@xena:/mnt/btrfs_ssd/iphonebackup# idevicebackup2 backup -u nnnnnnnn-nnnnnnnnnnnnnnnE .
.
<15 minutes later>
.
Removing 1 file
[==================================================] 100% Finished
Removing 1 file
[==================================================] 100% Finished
[==================================================] 100% Finished
Sending 'nnnnnnnn-nnnnnnnnnnnnnnnE/Status.plist' (189 Bytes)
Sending 'nnnnnnnn-nnnnnnnnnnnnnnnE/Manifest.plist' (193.3 KB)
Sending 'nnnnnnnn-nnnnnnnnnnnnnnnE/Manifest.db' (28.0 MB)
Received 13790 files from device.
Backup Successful.
root@xena:/mnt/btrfs_ssd/iphonebackup#
```

I can also tether to the phone, in the sense of using the iPhone as the "Wired connection 2" internet access device.

However, I should here note that, in the course of my investigative tinkering, I did create the new user line as prescribed by marav  (maravtdm, [](https://github.com/maravtdm)) 

(at the end of /etc/passwd)

`usbmux:x:52:83:User for usbmux daemon:/var/empty:/bin/false`

and I did build the patched version of usbmuxd-1.1.1 as prescribed by marav in post # 26 of

[](https://www.linuxquestions.org/questions/showthread.php?p=6496708)

I don't know whether or not the above two steps were crucial to success, or whether just leaving the stock Fedora 41 libimobiledevice rpm's, as installed by dnfdragora, would've sufficed.