libimobiledevice/libimobiledevice #1657: Release a new version

AsciiWolf, 2025-04-04:
The current last release (1.3.0) is from 2020 and there were many changes and fixes since then. For example the [SHA256 signature](https://github.com/libimobiledevice/libimobiledevice/commit/41543791dc9fa4d051305bf21000309f2dd47883) fix that is important for newer iOS releases.

Please, consider releasing a new version of libimobiledevice.

Thanks!

francis-drake, 2025-04-06:
SR: 2025-04-06 18:31:25.237903+0000 asr[28:788] [User Defaults] Could not connect to cfprefsd. No CFPreferences/NSUserDefaults changes will be saved

AsciiWolf, 2025-06-06:
@nikias Any chance the new version could be released? The latest one is from 2020 and is broken. Thanks!

AsciiWolf, 2025-06-06:
Oops, I have just noticed that the most important fix is still unmerged:

https://github.com/libimobiledevice/libimobiledevice/pull/1616

AsciiWolf, 2025-06-07:
Thanks! The fix from #1616 is already in some downstream distributions and confirmed to work as intended.

AsciiWolf, 2025-06-14:
I don't want to sound annoying or rude for asking one more time, but is the new libimobiledevice release planned to happen in the near future?

nikias, 2025-06-14:
Yes

AsciiWolf, 2025-06-18:
Feel free to ping me (via this ticket or email) if you need any help testing the new libimobiledevice version before its release. My possibilities are limited since I don't have any iOS device here at the moment, but I can try compiling the project on Debian and Fedora, and do some basic sanity testing.

AsciiWolf, 2025-08-09:
Any news about the new release? :-)

AsciiWolf, 2025-09-02:
Any news about the new release?