libimobiledevice/libimobiledevice #1667: iOS 18.5: idevicesyslog output is truncated

pekn, 2025-05-21:
With iOS 18.5, idevicesyslog output lines are truncated. This problem does not occur with iOS 18.4.1 or older.
So Apple has changed behavior, problem most likely isn't in idevicesyslog program.

Does anyone know if it's possible to get non-truncated output? If yes, how?

I tested with Apples logging profile (https://developer.apple.com/services-account/download?path=/iOS/iOS_Logs/loggingiOS.mobileconfig), but this problem occurs even with that profile installed.

pekn, 2025-06-24:
Sure, I will test soon.

nikias, 2025-06-27:
Could you provide some examples of truncated messages? I tried with an iPad on 18.5 and the log messages do not seem to be truncated. Just FYI, you can get the old behavior of idevicesyslog with `idevicesyslog --syslog-relay`.