libimobiledevice/libimobiledevice #1669: Altitude change set GPS location

BjornSmeetsYPTF, 2025-05-26:
I'm currently reviewing the code I used in my tool. Everything works well, but I have a question about location control—specifically, whether it's possible to simulate altitude or speed in addition to latitude and longitude. Based on what I've read in the [Apple Core Location documentation](https://developer.apple.com/documentation/corelocation/cllocation), I'm unsure if this is supported. Could someone clarify if there's a way to simulate altitude or speed, or is only latitude and longitude controllable through the available APIs?

coder-woo, 2025-08-06:
Hi, guy Which version did you use to test this? iOS 17+?

BjornSmeetsYPTF, 2025-08-06:
iOS 18

coder-woo, 2025-08-06:
[image]

sure？view source code does not support iOS 17+