libimobiledevice/libimobiledevice #1670: Minimum necessary free space on iDevice to perform backup

Melab, 2025-06-03:
I have an iPhone that is low on storage. Using iTunes and iMazing, the backup never completes. I noticed in one of the other issues in this project that changing the timeout length can effect this. Is this accurate? Would a longer timeout allow a backup of a device with little free space to complete? If so, how do I change the timeout?

mexmer, 2025-06-03:
if there is not enough space on iphone, it will fail to backup regardless timeout, increasing timeout only helps, when you have lot of files to backup, so iphone takes too long time, to calculate backup size, usually it's when people have tenths or hundred of thousands of photos in phone.

if backup fails because of low space, you need to make some free space, there is no other option. usually you can offload some applications, that will uninstall application while preserving application data. data of offloaded applications are stored in backup.

mexmer, 2025-06-09:
> > if there is not enough space on iphone, it will fail to backup regardless timeout, increasing timeout only helps, when you have lot of files to backup, so iphone takes too long time, to calculate backup size, usually it's when people have tenths or hundred of thousands of photos in phone.
> > if backup fails because of low space, you need to make some free space, there is no other option. usually you can offload some applications, that will uninstall application while preserving application data. data of offloaded applications are stored in backup.
> 
> And that amount of enough space is what?

that knows only apple engineers. some files (like photos) are backed up directly, while some files (like system configuration, call and message database and so on), are firt prepared internally for backup and stored in backup folder inside phone, before they are transfered, unforturnately there is no public documentation how this works exactly.

sometimes device will send back error at initial stage of backup, if there is not enough space either on source device or target device (eg. your computer), but in many cases it just internally crashes, and for you it looks it's stuck in endless wait.

like i said, check device system log, if you can find any error message related to backup. source will be either backupagent or mobilebackup2 service.