libimobiledevice/libimobiledevice #1672: Inconsistent / Bad C type usage

tihmstar, 2025-06-12:
Throughout pretty much all libimobiledevice projects, C type usage is somtimes inconsisten or bad/deprecated.

# The problems
The most prominent example are:
1) Using `unsigned char*` or `const unsigned char*` for data buffers instead of `void *` or `const void*`
This causes modern compiler (using modern C standards) to generate warnings of wrong type when using buffers of structured data.
Consider the following example with `plist_t plist_new_data(const char *val, uint64_t length);` as defined/implemented here https://github.com/libimobiledevice/libplist/blob/cf5897a71ea412ea2aeb1e2f6b5ea74d4fabfd8c/src/plist.c#L514-L522.
```
#include <plist/plist.h>

typedef struct {
  uint32_t someMagic;
  uint32_t someFlags;
  uint8_t someBytes[0x10];
} t_someStructure;

int main(void){
  t_someStructure s = {
  .someMagic = 0x123,
  .someFlags = 0xff,
  .someBytes = {},
  };
  plist_t test = plist_new_data(&s, sizeof(s));
  return 0;
}
```
Running `clang -I/usr/local/include a.c -c` yields:
```
a.c:15:33: warning: incompatible pointer types passing 't_someStructure *' to parameter of type 'const char *' [-Wincompatible-pointer-types]
   15 |   plist_t test = plist_new_data(&s, sizeof(s));
      |                                 ^~
/usr/local/include/plist/plist.h:261:50: note: passing argument to parameter 'val' here
  261 |     PLIST_API plist_t plist_new_data(const char *val, uint64_t length);
      |                                                  ^
1 warning generated.
```
While it is possible to silence the warning by using casts as in `plist_t test = plist_new_data((char*)&s, sizeof(s));` or `plist_t test = plist_new_data((const char*)&s, sizeof(s));`, i would argue that this is more of an API problem rather and a user problem since that wouldn't happen if the definition was `plist_t plist_new_data(const void *val, uint64_t length);` or even better `plist_t plist_new_data(const void *val, size_t length);` (the size_t thing i will discuss in a second seperately).

I would argue that since the intention is to pass "raw data", the underlying function shouldn't require `char*` type especially since it's not even required inside the function anyways and is only an arbitrary restriction posed by `plist_new_data` itself.

Now, i understand that sometimes it is useful to have a byte-iterable type _inside_ the function that handles the raw data in order to easily access offsets or increment the pointer. An example of this would be: https://github.com/libimobiledevice/idevicerestore/blob/a5905b7f905fc3cc83033ebd963f0dcba071e512/src/img3.c#L69
However, since the user of the function doesn't know (and doesn't need to know) the underlying structure, the user should still be able to handle the data as `void*`. Thus i suggest that the function should expect an unstructured `void*` (or `const void*` in this case) from the outside and then cast it internally. For example like this:
```
static img3_file* img3_parse_file(const void* data_, unsigned int size) {
const unsigned char* data = (const unsigned char*)data_;
....
```

2) Inconsisten usage of `int`, `unsigned int`, `uint64_t`, `uint32_t` for in-memory buffer sizes, where `size_t` should have been used.
Whenever a buffer is allocated in memory, it's size should be represented as `size_t` (or `ssize_t` if error information with negative numbers is needed) because a) it tells the user that this type is a memory-buffer-size-type and b) it has the correct bitwidth depending on the system used. i.e. on a 64bit system, this is 64bit wide, while on a 32bit system it is 32bit wide. Thus, on all (sane) systems (which we use) it is able to correctly represent the amount of bytes a buffer allocated in (virtual) memory can hold. 32bit system cannot hold more than 4GB in a single buffer (actually much less), but 64bit systems can.
The other types **don't** change across systems. On (sane) systems `int`, `unsigned int` and `uint32_t` is always 32bit wide, while `uint64_t` is always 64bit wide regardless of whether we're dealing with a 32bit system or a 64bit system.
This is a great pitfall for various unexpected and potentially hard to debug bugs.
For example taking `plist_new_data` which is currently implemented as following:
```
plist_t plist_new_data(const char *val, uint64_t length)
{
    plist_data_t data = plist_new_plist_data();
    data->type = PLIST_DATA;
    data->buff = (uint8_t *) malloc(length);
    memcpy(data->buff, val, length);
    data->length = length;
    return plist_new_node(data);
}
``` 
(see https://github.com/libimobiledevice/libplist/blob/cf5897a71ea412ea2aeb1e2f6b5ea74d4fabfd8c/src/plist.c#L514-L522)
If we pass 0x100000001 as a length to `plist_new_data` it works on a 64bit system and allocates 4GB+1Byte, but on a 32bit system it gets silently truncated and allocates only 1 Byte, potentially causing problems further down the program.

The inverse of this problem can be found in `int ipsw_extract_to_memory(ipsw_archive_t ipsw, const char* infile, unsigned char** pbuffer, unsigned int* psize)` https://github.com/libimobiledevice/idevicerestore/blob/a5905b7f905fc3cc83033ebd963f0dcba071e512/src/ipsw.c#L616. 
Since `unsigned int* psize` is a pointer to a 32bit number even on 64bit systems, this function can never extract (and correctly report to the caller) data into a memory buffer which is larger than 4GB. Given that modern IPSW files are already almost 10GB it is just a matter of time until one single file inside the IPSW will be larger than 4GB. Honestly, i'm surprised it hasn't happened yet. Or maybe it did and already causes some memory corruptions deep down in some programs.

[... 51 lines cut ...]
```
size_t size = 0;
....
struct zip_stat zstat;
....
size = zstat.size;
buffer = (unsigned char*) malloc(size+1);
```
A ZIP is a file on disk and can be larger than 4GB even on 32bit systems. A file inside a zipfile can also be larger than 4GB.
`struct zip_stat` defines size as `zip_uint64_t size;`, which really is a `uint64_t`. Thus by trying to extract a 4GB+ file on a 32bit system, the line `size = zstat.size;` will truncate the size to only 32bit and allocate a smaller buffer. In this case there probably isn't a memory corruption, but it still won't function correctly and the messages further down will only print cryptic errors due to truncation.
Maybe this isn't the best example because a 4GB file cannot be loaded to memory on 32bit systems anyways, but i can think of other places where large files are handled on 32bit system where the data doesn't fully reside in memory where using `size_t` instead of `uint64_t` would be problematic.

# Recommendation
Having explained the problems, here are my recommendations for API change (both internally facing and externally facing):
- use `const` wherever the callee doesn't modify the date in the buffer (libimobiledevice already does a pretty good job with this)
- use `void *` in function declaration whenever the callee doesn't require the caller to provide data with structure of any kind. (for example with `plist_new_data`)
- use `size_t` when describing a in-memory buffer size.
- use `uint64_t` when describing a on-disk file size.

# Final remarks
I understand that the origin of libimobiledevice code is really old and back in the days these problems weren't yet a thing and i understand people back then were less aware and cared less about this. GCC compilers maybe didn't warn you about type mismatch and 64bit vs 32bit systems weren't as big of a deal, yet a consideration. However, nowadays our compilers better help us to write better code and warn us about type mismatch and 64bit systems are common and 32bit systems are sometimes still used. In order to avoid all kinds of problems now and in future i think it's time to address these issues, update the APIs and rewrite some code to care for correct types in the right situations. It is a good opportunity to clean up before having to debug real memory corruption issue in future.

Fixing issue 1) with `char*` -> `void*` shouldn't impact any existing code and fixing issue 2) with the sizes will only cause compilation issues when there already is a bug. I suspect that addressing issue 3) is probably only (or mostly) needed internally and not on API level.

Maybe it's worth enabling `-Wall -Werror` in debug builds or as a special configuration to help clean up the code and get rid of all the warnings. But please **DO NOT ENABLE THIS IN NORMAL BUILDS** as this will just break compiling source code altogether when the next compiler update ships.

nikias, 2025-06-12:
Thanks for the report. Did you write all that manually? Damn... anyway I am aware of the mess and will try fixing it. Already fixed the ostrace API I posted yesterday.

tihmstar, 2025-06-14:
Yes, i did write all of that manually. Calling this AI generated is mean 😢

nikias, 2025-06-14:
> Yes, i did write all of that manually. Calling this AI generated is mean 😢
I was just surprised with the effort you put into this. I think API wise in terms lf code compatibility it should not break if I change sizes of parameters, it will just not be binary compatible here and there, so as long ss the library version (soversion) is updated it shouldn't be an issue or am I missing something?

nikias, 2025-06-23:
I think I need to push out a new release first, and then start in order with libplist, libusbmuxd, etc. The thing is that it breaks ABI when changing sizes, but I also don't really want to have annoying multiple versions of functions (like `afc_file_read2` or similar). I'd rather change them and hope for not too many complaints...

tihmstar, 2025-06-28:
It's an unfortunate effect of having to update legacy code, but honestly i think it's worth it.
I think it's reasonable to bump major version for a "big" change like that, then have maintainers handle (in)compatibility.
In the long run there is benefit outweights the extra effort that'll need to be put in once.
And i do agree that adding more functions like `afc_file_read2` is not worth it.

nikias, 2025-06-30:
Regarding `ipsw_extract_to_memory`. I can change the signature but the other problem remains. Honestly that function was never supposed to allow larger files because it ends up eating too much memory. All large files are either streamed directly or extracted to disk. So I guess on a 32 bit platform it will just have to error out if the file size (zstat.size) is larger than 32 bits unsigned.

tihmstar, 2025-07-26:
Do you want me to help you with the API changes? I.e. start with libplist and create a pull request.
Or are you "finishing up" things first and are gona get to it afterwards?

mexmer, 2025-07-28:
> Regarding `ipsw_extract_to_memory`. I can change the signature but the other problem remains. Honestly that function was never supposed to allow larger files because it ends up eating too much memory. All large files are either streamed directly or extracted to disk. So I guess on a 32 bit platform it will just have to error out if the file size (zstat.size) is larger than 32 bits unsigned. But as you said, this isn't the best example function anyway.

what library you use for extraction? libraries i use to work with archives for many years support platform independent 64bit file size, and i don't mean just archive, but also files inside archives, so only issue with file size over 32bit max is when underlaying filesystem doesn't support it ... which is independent on architecture (for example FAT32, or old HFS version 1), so if you either stream, or extract to disk, there should not be problem with 64bit filesize on 32bit system.

as for ipsw_extract_to_memory, that function should not be used for large ipsw files anyways, so i don't see problem with having it limited to 32bit.
if you change it to 64bit, just check if size at entry, and bail out, on 32bit system, since memory manager will not allow such allocation anyways.
on some operating systems this allocation can fail even on 64bit system, if there is no contiguous block of memory available for allocation (Eg. if there is high memory fragmentation). you can see this offten on windows, but i've encoutered it even on 64bit linux, i think they did some improvement in memory management since 4.20 kernel tho', so it might not happen on modern linuxes.