libimobiledevice/libimobiledevice #1674: Problem with Photos update

lbdann2, 2025-06-18:
I'm trying to send a file with my video/photo to DCIM so it will show up in the gallery, but it's not happening.

Through 3utools I was able to send the photo to the gallery, but it doesn't work with the video, it just doesn't show up in the gallery. Maybe someone knows the solution to this problem? How can I make the videos I send to my Iphone appear in the gallery?

mexmer, 2025-06-18:
videos needs to be in correct format, if i'm not mistaken, it needs to be encoded with .h264 or .h265 codec and stored as heiv or mp4 container. still no idea, if this will work with current ios version, i remember mp4 .h264 was working in past, but i didn't use direct insert probly since ios 11. you should probly ask 3utools support for help, they should know what format needs to be used, if 3u tools supports uploading video.

libimobile doesn't include functionality to modify photos and videos library in ios device.

lbdann2, 2025-06-19:
> videos needs to be in correct format, if i'm not mistaken, it needs to be encoded with .h264 or .h265 codec and stored as heiv or mp4 container. still no idea, if this will work with current ios version, i remember mp4 .h264 was working in past, but i didn't use direct insert probly since ios 11. you should probly ask 3utools support for help, they should know what format needs to be used, if 3u tools supports uploading video.
> 
> libimobile doesn't include functionality to modify photos and videos library in ios device. it allows you to access part of files trough AFC service, inserting photo or video into iOS device, requires additional steps that are not part of library.

Perhaps you know what it takes? I'm using pymobiledevice3, via libimobiledevice worked too. The video is in the correct format, loads into memory, but doesn't show up in the gallery. That said, I know Private software that quietly does this, but I don't know how.