libimobiledevice/libimobiledevice #1683: No such file or director error

aknight2015, 2025-08-13:
I follow the directions on your download page to the letter, and get the above error. Here's a screenshot to help.

[image]

mexmer, 2025-08-13:
no, you didn't follow instructions, because you are trying to run autogen.sh inside your home folder, while it needs to be executed inside folder, where source code is.

aknight2015, 2025-08-13:
I understand source code compilation as most of you developers require linux users to compile your products. What you fail to understand is that you simply say "Source" without specifying which. Do I need to do it for every single one going down that list? Do I need to compare the list to the list of dependencies being installed to see which one I need to compile?

mexmer, 2025-08-13:
most common linux distributions already contain binaries of libimobile and tools so you don't need to compile from source code.

names of packages are either same or similar, so instructions for ubuntu works for most distributions, that utilize APT for managing packages.
rpm, yum, yast ... and other package manages, all have install command, that you can use for installing binary packages

you only need source code compilation in 2 cases
1) your linux distribution doesn't have binary packages for libimobiledevice at all
2) your linux distribution has packages based on outdated code

when you build from source code, you should remove all outdated binary packages

when you build from sourcecode, you need to compile every source that is listed on website, simply to have all components up to date, otherwise it might fail to compile (better case), or introduce instalatibility and incompatibility (worse case)

order is important, so you should follow this order
  libplist
  libimobiledevice-glue
  libusbmuxd
  libimobiledevice
  libirecovery
  libtatsu
  libideviceactivation
  usbmuxd

optional components, if you want to use them, compile last, in any order
  idevicerestore
  ideviceinstaller
  ifuse

there is also script made by @nikias , that will download sources and compile them for you
https://is.gd/limdlinux
but i think, it's outdated, because it doesn't download and compile libtatsu

aknight2015, 2025-08-14:
I don't think I've had a gentler chastisement in my whole life. Yes, I want to see a new birth of freedom as well. For all mankind.